wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #1: Turn the Grove Vision on/off

victor-wildlife, 2024-06-07:
Developed in the forked https://github.com/wildlifeai/Seeed_Grove_Vision_AI_Module_V2

It might be worth checking other forked repositories to find out if others have already done it. Maybe (https://github.com/HimaxWiseEyePlus/Seeed_Grove_Vision_AI_Module_V2/commit/d23da85936fa23f9eef527b02ed6dcfe69bcdbd2)?

victor-wildlife, 2024-06-07:
@Tobyntobyn any updates on this?

Tobyntobyn, 2024-06-08:
@victor-wildlife Yeah have been making progress with the limited time available lately! 
The device is switching between low power and high power modes with a timed interrupt. 
I'm currently progressing towards an image being exported when "on" state, having a few compiling issues with initialisations. 

As I'm building from the example SDK programs, this will have the person detection integrated into the program which I'm hoping to exchange for our rat detection model (I got access to the Edge Impulse project) once I get past the current hurdle.