wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #16: Update the yaml file to dinamically load the right compiler version

victor-wildlife, 2024-11-19:


victor-wildlife, 2025-01-24:
@Tobyntobyn Is this something that Dev or Tim could help us with?

Tobyntobyn, 2025-02-14:
There is a branch (dynamically-select-compiler) for proof of concept that we can do this. On execution, this might not be the best solution for our project as the most recent version runs into compile time errors. 
I've been aware of it for awhile but haven't properly considered it until now, but shifting the pathway to look into docker containers instead. This enabling easier environment set up for new developers and environment consistency between the team.

victor-wildlife, 2025-03-02:
@Tobyntobyn is this related to the work Ash and you have been doing?

ashceniza, 2025-03-11:
@victor-wildlife  @Tobyntobyn , I've not looked at this issue before or that branch, however I have docker'ised this (for want of a better word).

I think I need to set up a call with Tobyn and show him what I've done and find a way to incorporate this in your workflow, I've done it independently and I have not created a branch as such.  

After the conversation Tobyn we can work out way that seamlessly fits in your existing workflow, I don't want to disrupt your existing flow without understanding the consequences of the present state, it will just set things back... of course we need to take the long term view as well, by that I mean for new developers coming on , how do we set up a QuickStart environment by leveraging "conternarisation" where you have a repeatable solid every time, take the configuration nightmare out of the loop, this is the advantage of using docker.

Currently my environment looks like  so, notice I've cloned two separate branches and can get either one up and running very quickly.

[image]

[image]

karolinakisiel, 2025-04-16:
@ashceniza @Tobyntobyn - would the docker containerization take care of this?

karolinakisiel, 2025-04-22:
@ashceniza @Tobyntobyn - see my question from last week.

Tobyntobyn, 2025-04-22:
@karolinakisiel Yes correct, this has been addressed and resolved within the docker container. We explicitly select the compiler version within the container. This is a good thing as development happens relative to a version, but as new versions get released, it is easy to test and making respective changes to our firmware.

closed 2025-04-22 by Tobyntobyn