wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #27: Catologue/Document Himax's makefiles

victor-wildlife, 2025-02-09:


Tobyntobyn, 2025-05-13:
Haven't look into this yet. I'm prioritizing the other firmware tasks that MVP2 is relying on for now.

Tobyntobyn, 2025-06-12:
@karolinakisiel hasn't been done yet. The basic understanding is there to understand what each Makefile does and how it's required. But not to the extend to know how to bring out project independent from their SDK.  

For now, there are 4 Makefiles used per program compiled within the SDK. 

- The top level Makefile that sets the compiler location, which top level engine's and libraries to use, and relevant for us, which "project_folder" to use (were we define ww_projects).
- app.mk which is used to set application source directories, for our project setting the location to "$(EPII_ROOT)/app/ww_projects/ww.mk"
- ww.mk, this makefile is contained within our ww_projects direcotry, it sets extra macros, but this is primiliarly relevant to us to select which program we want to compile (eg. ww130 or ww500). 
- program.mk (eg. ww130_cli.mk), this is the program specific makefile, used to set the relevant libraries specific for the program getting compile, sets the camera/sensor that is being used (eg. RP v1, v2, etc.), pointers to specific linker script files, etc. 

There is work that can be put in to optimize the compile times, make our projects independent and also try to step away from using so many makefiles.

karolinakisiel, 2025-07-22:
@Tobyntobyn  can i set this one to done?

Tobyntobyn, 2025-07-23:
@karolinakisiel Keep this open until we hear reply from Steve as Victor asked:
As our projects grow and we onboard more environmental partners to use the Wildlife Watcher, we've identified that the current GitHub codebase structure (mixing examples with project code) could pose challenges for updates and customer adoption. If possible, we would love to discuss a revised structure where your SDK resides in a dedicated Seeed Grove Vision AI Module V2 folder, and our projects (and those of other customers) are in a separate Workspace folder. 
Title of email: Wildlife Watcher updates