wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #28: Branches cleanup - February 25

victor-wildlife, 2025-02-09:


victor-wildlife, 2025-02-09:
@Tobyntobyn can you please outline the steps we need to follow to do a little bit of a cleanup of the 7 branches?

Tobyntobyn, 2025-02-11:
The different branches currently have some core functions we want to include in our project, and two failed exif branches of mine (tinyexif & EXIF). 
I have kept these two exif branches for now as a reference for the developer that takes over this task. Showing them two methods attempted for exif handling (i.e what not to do).

The remainder of the branches are as follows:
- WW500: Specific to the new WW500 board, this is currently in development. This branch cannot be merged into main yet as I don't have one of these board. I need the WW130 specific code for me to continue developing as well
- timer-delay: Completed feature for adding an interval to captures. Ready to merge into main.
- image-filename-update: Export image folder structure set with file naming convention added. Completed feature and ready to merge into main

Tobyntobyn, 2025-02-11:
I haven't created pull requests for the two completed branches yet as I was wanting to sync up with Charles first to see his status on the WW500 branch. I want to pull his branch up-to-date with the Himax updates that we merged in a couple weeks ago, double check there weren't merge conflicts from that rebase _before_ including the extra features that I've written.

victor-wildlife, 2025-02-21:
@Tobyntobyn what is the status on this branches cleanup?

Tobyntobyn, 2025-02-24:
Completed and up-to-date! Charles has the OK for his DFU work, the timer interval is successfully running on the WW500 board, and the updated folder structure is merged into main. 
Apologies for not closing this issue.