wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #29: Dinamycally load new ML models to Grove Vision

karolinakisiel, 2025-02-10:


karolinakisiel, 2025-04-15:
@victor-wildlife - this is now Beta phase rather than MVP2, correct?

Tobyntobyn, 2025-07-22:
This task can be broken down into two stages:

1. Mobile app --> SD card
2. Initialise model from SD on boot

For the first stage, for MVP2 and Beta phase we can assume the user can place the model on the SD card via their computer as we want to prioritise the second stage. When we develop the mobile app load feature, we want the user to select their model (potentially to have direct authentication to their Edge Impulse account) and on button press, for it to load onto the SD card via BLE when connected to the target device. 
The user will also need confirmation their model has been successfully uploaded and is working. Some feature than enables them to have the option to upload an test image to the WW device and get the results returned to them on the mobile app (lets flesh this out via task #51 "Brainstorm pipeline for user to test their custom model on deployment")

The second stage is to load a model on boot directly from the SD card, instead of loading it via Xmodem UART protocol or having it embedded inside the firmware. The general approach for this is to first initialise fatfs on boot _before_ the cv_init function gets called. Once the correct directory is known, to then use this location to initialise the model instead of the location in flash memory. 

The last piece (I might as well include this here for reference), is the part from Edge Impulse --> Mobile app. The model needs to be formatted as a model_vela.tflite model, and we need to find a solution to make it as easy as possible for a user to access this model within our mobile app. As I mentioned above, if we can find a method the have auth directly to a users Edge Impulse account, this would avoid our users downloading their model locally and having to re-upload it to Supabase

closed 2025-12-24 by victor-wildlife