wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #30: Save output of the ML models to EXIF

karolinakisiel, 2025-02-10:
Finding an app that works and saves the 
insert some EXIF information

Tobyntobyn, 2025-04-29:
Current trained rat detection model results are:
not rat 
rat 
uncertain 
Results for there are out of 1 (eg. rat 0.79, not rat 0.21)

We could look at inserting the person detection model results into EXIF for now and update once we convert to the rat model. Eg for person detection results: person_score = 92 no_person_score = -92 

- [ ] Insert person detection results into the APP1 block
- [ ] Update EXIF config files so exiftool readers produce interpretable results
- [ ] Configure ML model for SDK requirements
- [ ] Replace person detection model with rat model
- [ ] Update APP1 block & config files for rat detection results

Tobyntobyn, 2025-05-13:
I have a pull request for the EXIF insertion into main. Once this request gets reviewed/accepted (By Charles), I'll update my rat detection model branch to sync up with the EXIF work, and can continue on with the remainder of the tick boxes above.
Until we merge, the two 'features' are independent from each other

Tobyntobyn, 2025-05-26:
All done on branch edge-impulse-rat-model. Not merged with main yet. 
Still untested, dynamic model loading using the same address in memory (i.e. load a variety of different models without updating memory addresses)

closed 2025-05-26 by Tobyntobyn