wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #36: Review the compiler dependencies

victor-wildlife, 2025-02-25:


ashceniza, 2025-02-25:
1. Current setting up a docker image to compile the code as documented.  I'm building it on a Mac so will be using the ARM arm-gnu-toolchain-13.2.Rel1-aarch64-arm-none-eabi.tar.xz in a docker container. 
2. Will compile using a known version to ensure we can get a clean build after this is complete I will use the latest version to find out the errors and report back, or hopefully resolve the issue.

ashceniza, 2025-02-25:
- I have it compiling in the container, I will leave it for while...., the next step is to generate the image after compilation, will do that tomorrow.

ashceniza, 2025-02-26:
Tobyn and I have had conversations regarding compilers... We can close this issue as he is happy with ocntainerisation

closed 2025-02-26 by ashceniza