wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #37: Select method for device keeping timedate

Tobyntobyn, 2025-02-25:
The HX6538 does not have an integrated RTC (real time clock) so need to have a method to keep timedate.

From chatGPT, options to decide on what methods we should use for this are:
1. Use an External RTC Module (Best for Accuracy)
  You can connect an I²C or SPI RTC module like:
  DS3231 (highly accurate, has temperature compensation)
  PCF8563 (low power)
  RV3028 (ultra-low power)
  The RTC keeps time even when the HX6538 is powered off (if it has a backup battery).
2. Synchronize Time Over I²C or UART
  If your I²C master (e.g., Raspberry Pi, another MCU) has accurate time, it can periodically send the current timestamp to HX6538.
  Use a timestamp-based approach (e.g., Unix time).
3. Approximate Time Using FreeRTOS System Tick
  If you only need relative time since boot, use xTaskGetTickCount() in FreeRTOS.
  However, this resets on reboot and may drift over time.
4. Use a GPS Module for UTC Time (If Available)
  If your HX6538 device is outdoors, you can use a GPS module (e.g., NEO-6M, NEO-M8N) to get an accurate UTC timestamp.
-----------------------------
I used a timestamp apporach "#define EPOCH_TIME_BASE 1735689600" that was in theory the timestamp for Jan 1 2025 00:00:00, however on execution it was "...". Short answer is I don't trust the timestamp approach.

Our SDK currently doesn't recognize the xTaskGetTickCount so will need to include another library, and also requires an initial timestamp (I think).

acutetech, 2025-02-25:
I assume the objective here is to add a timestamp to images in the EXIF. Right? If so the best way to do this would be to ask the nRF52832 for the time, either from time to time, or every time it is required.

These sorts of processors typically have a number of clock sources - for example:
1. A crystal oscillator running at something like 24MHz that runs the processor core and peripherals. When running it is pretty accurate (e.g. 50ppm) but the clocks are typically shut off when the processor sleeps, to save power.
2. A crystal oscillator running at 32768Hz that runs always. This is used for accurate time-keeping including maintaining calender time and date (the RTC function). It is typically 20ppm.

The nRF52832 has 1 & 2 but the HX6538 does not have a 32768Hz crystal (it has pins for it - it is just not implemented on our board, and anyway there is no point in duplication). 

The nRF52832 module has a 32768Hz crystal. Its main purpose is to provide accurate timing for BLE and LoRaWAN radio events - e.g. to wake the processor to listen for incoming radio packets expected a fixed time after a radio transmission. It can also be used to keep UTC time.

The LoRaWAN spec allows the LoRaWAN servers to send accurate time information. When received the nRF52832 synchronises its time to the message received, and uses the 32768Hz clock to keep the time from that point on, until the next time is received from the LoRaWAN network.

I can make this time available to the HX6538 via the I2C link. Probably it is best if the HX6538 code requests it. Check out the [systime.h](https://github.com/Lora-net/LoRaMac-node/blob/master/src/system/systime.h) to see what is available - looks like you can have seconds and milliseconds (Unix time) or a time converted to a calender time of [gcc tm type](https://github.com/eerimoq/arm-toolchain-windows/blob/master/arm-none-eabi-gcc/4.8.3-2014q1/arm-none-eabi/include/time.h) which might be more convenient for EXIF. 

If you think this is the right approach then let me know and I can add something.

Of course, the WW500 will need an initial network message to set the time in the first place. We should probably implement an addition command so that the app can send the UTC time, in case the board has not got a LoRaWAN connection.

I strongly advise the use of UTC and not attempting a conversion to a local time zone. This could be done at later time, for presentation purposes, at a web page etc.

victor-wildlife, 2025-03-03:
copying previous message here. 
The HX6538 does not have an integrated RTC (real time clock) so need to have a method to keep timedate.

From chatGPT, options to decide on what methods we should use for this are:
1. Use an External RTC Module (Best for Accuracy)
  You can connect an I²C or SPI RTC module like:
  DS3231 (highly accurate, has temperature compensation)
  PCF8563 (low power)
  RV3028 (ultra-low power)
  The RTC keeps time even when the HX6538 is powered off (if it has a backup battery).
2. Synchronize Time Over I²C or UART
  If your I²C master (e.g., Raspberry Pi, another MCU) has accurate time, it can periodically send the current timestamp to HX6538.
  Use a timestamp-based approach (e.g., Unix time).
3. Approximate Time Using FreeRTOS System Tick
  If you only need relative time since boot, use xTaskGetTickCount() in FreeRTOS.
  However, this resets on reboot and may drift over time.
4. Use a GPS Module for UTC Time (If Available)
  If your HX6538 device is outdoors, you can use a GPS module (e.g., NEO-6M, NEO-M8N) to get an accurate UTC timestamp.
-----------------------------
I used a timestamp apporach "#define EPOCH_TIME_BASE 1735689600" that was in theory the timestamp for Jan 1 2025 00:00:00, however on execution it was "...". Short answer is I don't trust the timestamp approach.

Our SDK currently doesn't recognize the xTaskGetTickCount so will need to include another library, and also requires an initial timestamp (I think).

Tobyntobyn, 2025-03-06:
@victor-wildlife this raises answers towards the datetime handling once EXIF is embedded but unfortunately not towards the problem of the base problem of inserting EXIF data.

Charles is having a look at implementing the UTC command as it'll involve some firmware updates on the nRF module as well

acutetech, 2025-03-06:
I will do something similar for GPS location. At present I propose that the app sends the location with an AI gps <parameters> string and so the 'gps <parameters> will be delivered to the HX6538 code.  Copilot suggested that the <parameters> format is a bit complex. I will propose something that delivers a string through to the HX6538 that does not need to be parsed - just saved and copied into the EXIF. Or else the processing in simple.

I asked: what is the usual format for EXIF GPS coordiantes?

Copilot said:
EXIF GPS coordinates are stored in several tags within the EXIF metadata to record the latitude, longitude, altitude, and other related information. Here's the usual format for each of these tags:

1. **GPSLatitudeRef**:
   - Indicates whether the latitude is north or south.
   - Values: 'N' (North) or 'S' (South).

2. **GPSLatitude**:
   - The latitude in degrees, minutes, and seconds.
   - Format: Three unsigned rational values (degrees, minutes, seconds).

3. **GPSLongitudeRef**:
   - Indicates whether the longitude is east or west.
   - Values: 'E' (East) or 'W' (West).

4. **GPSLongitude**:
   - The longitude in degrees, minutes, and seconds.
   - Format: Three unsigned rational values (degrees, minutes, seconds).

5. **GPSAltitudeRef**:
   - Indicates the altitude reference.
   - Values: 0 (above sea level) or 1 (below sea level).

6. **GPSAltitude**:
   - The altitude in meters.
   - Format: A single unsigned rational value.

Here’s an example of how GPS coordinates might be represented in EXIF metadata:

```plaintext
GPSLatitudeRef: N
GPSLatitude: 37/1, 46/1, 2183/100
GPSLongitudeRef: W
GPSLongitude: 122/1, 25/1, 3060/100
GPSAltitudeRef: 0
GPSAltitude: 30/1
```

In this example:
- Latitude: 37 degrees, 46 minutes, and 21.83 seconds North.
- Longitude: 122 degrees, 25 minutes, and 30.60 seconds West.
- Altitude: 30 meters above sea level.

These values are stored as rational numbers, where each number is represented by a pair of integers (numerator and denominator). This allows for precise representation of the coordinates.

For more information on EXIF GPS tags, you can refer to the [ExifTool's GPS Tags documentation](https://exiftool.org/TagNames/GPS.html).

Perhaps I will truncate this, so the <parameters> look like this:

```
N,37/1, 46/1, 2183/100,W,122/1, 25/1, 3060/100, 0,30/1

```

Tobyntobyn, 2025-03-10:
For our exif field requirements, we only need to include two fields: latitude & longitude 
Any other variables we pass over would be for our own project's (or DOC's) interest.

victor-wildlife, 2025-03-11:
@acutetech and @Tobyntobyn my 2 cents, whenever it's possible we should try using the format and unit names from [Camtrap DP](https://camtrap-dp.tdwg.org/data/#deployments) and if not available those from Darwin Core standards ([Georeferencing](https://docs.gbif.org/georeferencing-quick-reference-guide/1.0/en/) and [time/date](https://techdocs.gbif.org/en/data-processing/temporal-interpretation#:~:text=A%20single%20day%20is%20represented,%2D18T13%3A27%3A00%20.)) TLDR location fields include should use the WGS84 datum and a "coordinateUncertainty" expressed in meters, timedate should use the [ISO 8601-1:2019](https://en.wikipedia.org/wiki/ISO_8601) format

acutetech, 2025-03-12:
AFAIK the EXIF spec is very specific about the format of the EXIF fields. In any event, it is saved as binary data, not ASCII strings. 

I think we must use the EXIF format for saving metadata inside the JPEG file. That is the way to have jpeg files that are compatible with the standard and can be used without surprises by others. If there is conversion to be done to a different format then this can be done later e.g. on a server.

EXIF specifies that a GPS coordinate (lat or long) is stored in degrees, minutes and seconds, and each of these is saved as  a pair of 32-bit words - one as a numerator and one as a denominator. These 6 words (24 bytes) can be  extracted and converted to a decimal number representing degrees, but at a later stage. (it also stors a reference 'N', 'S', 'E', 'W' as a string).

I have not looked (yet) at datum and uncertainty aspects.

acutetech, 2025-03-12:
I have just added an interim upgrade to the ww130_cli app. This is in the utc-gps branch now.

It adds a new exif_gps.c & .h which should allow much of the management of setting and retrieving the GPS location. This includes 3 test functions, which are implemented in check_GPS() in ww130_cli.c. I suggest you look at what this does and please try to find any errors. The several "DEBUG" messages can be removed in due course: these helped me debug.

There are new CLI commands 'setgps' and 'getgps' and I have done a little testing that shows that the GPS location can be set from the app.

There is a good deal of explanation at the top of the exif_gps.h file. Also, a link to a conversation I had with Chat GPT is [here](https://chatgpt.com/share/67d0aed1-2abc-8005-b2a4-2c755e47749a) - this goes into some detail about how to embed the stuff I have done into a JPEG file. Of course, there will be other ways to do this.

There are some initial steps towards getting the time set as well, but this is more complicated since the HX6538 will have to ask the MKL62BA for the UTC time (at least at present). (There is a local RTC in the HX6538 but it seems to be running 16000 times too fast...)

When Tobyn is happy then we can ask Miha to add commands in the app to set the GPS location and the UTC time.

acutetech, 2025-03-12:
PS - the ChatGPT conversation seems to have all my interactions with it. Most of teh earlier part I have incorporated in my code. The section starting "EXIF GPS Data Structure in a JPEG File" is the part that I have not implemented, and which might possibly be of interest to Tobyn if he is trying to do this.

Also - I guess I should create a corresponding file 'exif_utc.c' to encapsulate the UTC functionality.

victor-wildlife, 2025-03-21:
@Burzo is this task being tracked somewhwere else? If not, are you clear on the steps you need to follow to enable the exchange of timedate and GPS information?

acutetech, 2025-03-21:
No one needs to do anything at the moment. I am making good progress on a mechanism to get the app to provide both time and location. I will inform everyone when next steps are needed. A few days.

karolinakisiel, 2025-04-06:
@Tobyntobyn - can you confirm that it's merged and ready for @Burzo? thanks

Tobyntobyn, 2025-04-07:
Merged and relevant information passed onto @Burzo via two email threads:
New commands: setutc, setgps, sendphoto & JPEG file transfer to app (not directly corelated to UTC-GPS commands, but it goes in line with the method to achieve of setting the app up to make command calls via CLI structure)