wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #40: Develop the code behind the "Start Deployment" button

victor-wildlife, 2025-03-04:
Develop the code and approach to what happens after the mobile app sends the "start deployment" command.

victor-wildlife, 2025-03-11:
@acutetech has also started to work on this? I believe he would do a UML diagram. It will be great to include the link here

karolinakisiel, 2025-04-06:
@Tobyntobyn - just checking on progress of this. thanks

victor-wildlife, 2025-04-06:
According to a previous list, these were the steps required

- [ ] Review Miha's DFU functionality.
- [ ] Continue developing the "DFU Extensions" software.
- [ ] Use the core code for reliable data transfer via BLE.
- [ ] Add functions to transfer files in both directions (jpeg).
- [ ] Add this to the WW500 and test it.
- [ ] Miha needs to add it to the app.

Tobyntobyn, 2025-04-07:
I agree with these steps. I don't have too much direct involvement to achieve these as I don't have a WW500, Charles is tackling the DFU extension, and Miha is all over the app! :)

ashceniza, 2025-06-15:
@karolinakisiel  @victor-wildlife ,  I don't have the boards WW500  also my bandwidth is limited, at the moment but will let you know if things have changed and I could do this piece of work which looks interesting :-).  Also I'll have to speak to @Tobyntobyn and @acutetech  to see what state this issue is at present.