wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #43: Dockerise SDK

victor-wildlife, 2025-03-05:
Steps to containerise the SDK.

- [ ] Fix the compilers to remove compiler warnings

- [ ] Make it work under the 14.2 version

ashceniza, 2025-03-07:
Charles has fixed the compile issue under 14.2, the fix is in the utc-gps branch of the code, I suppose Charles will test and then push it to the main branch when its fixed.

ashceniza, 2025-03-07:
```bash
file path = /app/src/we2_image_gen_local/output_case1_sec_wlcsp/
boot_plain              /app/src/we2_image_gen_local/output_case1_sec_wlcsp/hx_layout_ota.bin
boot                    /app/src/we2_image_gen_local/output_case1_sec_wlcsp/hx_layout_ota.bin
secboot_cert_file       /app/src/we2_image_gen_local/output_case1_sec_wlcsp/sign_formal_hx_layout_ota.bin
boot list length = 1
['/app/src/we2_image_gen_local/output_case1_sec_wlcsp/hx_layout_ota.bin']
exit status code: 0
signed (or raw sec type) file name = sign_formal_hx_layout_ota.bin
AddFlashAlignPadding
Output image: output_case1_sec_wlcsp
Output reference data: output_case1_sec_wlcsp/layout_ota.xml
Output reference data: output_case1_sec_wlcsp/layout_ota.json

GEN_WHOLE_IMAGE
flash address = 0x14000, len of cur data = 0
img = /app/src/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_bootloader.bin
flash address = 0x1000, len of cur data = 163840
img = /app/src/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_2nd_bootloader.bin
flash address = 0x0, len of cur data = 163840
img = /app/src/we2_image_gen_local/output_case1_sec_wlcsp/sign_formal_hx_layout.bin
flash address = 0x28000, len of cur data = 163840
img = /app/src/we2_image_gen_local/output_case1_sec_wlcsp/sign_formal_hx_layout_ota.bin
flash address = 0x29000, len of cur data = 167936
img = /app/src/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p1.img
flash address = 0xAB000, len of cur data = 700416
img = /app/src/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p2.img
Output image: output_case1_sec_wlcsp/output.img
Output image: output_case1_sec_wlcsp/output.img

IMAGE GEN DONE
root@9b92ac0fe13e:/app/src/we2_image_gen_local# 
```

closed 2025-03-11 by ashceniza