wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #47: Dockerize the project

Tobyntobyn, 2025-03-21:
Get consistency among environments by wrapping our project in a docker container.

ashceniza, 2025-03-22:
@Tobyntobyn 

Started working on it this weekend and have managed to complete it, now I have to document the thing and then push up the dev branch, so you can test.

Below is the container being built and then following snippet is the image being generated. 

The process is driven by a Makefile to ,
1. Build the container and map the project (got clone Seed..<branch_name>) to the container  "make dev-build dev-up'
2. Once the container is up and a terminal to the container is up, just run the script genimage.sh to clean, compile and generate the image as per documentation. 

```bash
 make dev-down dev-build dev-up
docker-compose -p waidev    down development
INFO[0000] Any of the services [development] not running in project "waidev" 
/Applications/Xcode.app/Contents/Developer/usr/bin/make rm-image || true
docker             rmi waibase:vnext    
Untagged: waibase:vnext
Deleted: sha256:e6e963c8490d76c00e42924e526eb55765bd20b0fecd1c783cbd83f33031a9bb
time docker-compose -p waidev    build --no-cache development    
Compose now can delegate build to bake for better performances
Just set COMPOSE_BAKE=true
[+] Building 203.9s (14/14) FINISHED                                        docker:desktop-linux
 => [development internal] load build definition from Dockerfile.base                       0.0s
 => => transferring dockerfile: 1.51kB                                                      0.0s
 => [development internal] load metadata for docker.io/library/ubuntu:22.04                 1.6s
 => [development internal] load .dockerignore                                               0.0s
 => => transferring context: 89B                                                            0.0s
 => [development 1/8] FROM docker.io/library/ubuntu:22.04@sha256:ed1544e454989078f5dec1bfd  0.0s
 => => resolve docker.io/library/ubuntu:22.04@sha256:ed1544e454989078f5dec1bfdabd8c5cc9c48  0.0s
 => [development internal] load build context                                               0.0s
 => => transferring context: 423B                                                           0.0s
 => CACHED [development 2/8] WORKDIR /opt                                                   0.0s
 => [development 3/8] COPY packages.txt /tmp/packages.txt                                   0.0s
 => [development 4/8] RUN apt-get update &&     xargs -a /tmp/packages.txt apt-get instal  20.5s
 => [development 5/8] RUN wget --progress=bar:force:noscroll https://developer.arm.com/-  141.2s
 => [development 6/8] COPY entrypoint.sh /usr/local/bin/                                    0.1s 
 => [development 7/8] COPY genimage.sh /usr/local/bin/                                      0.0s 
 => [development 8/8] RUN chmod +x /usr/local/bin/entrypoint.sh &&     chmod +x /usr/local  0.2s 
 => [development] exporting to image                                                       40.1s 
 => => exporting layers                                                                    33.1s 
 => => exporting manifest sha256:3fb0224e935367f36088df164488e0aca1ffcaab607bf5542b6e7156c  0.0s 
 => => exporting config sha256:ace4a5155f646adad054eac06d67e3fe6d037ee7862e6bbaa74b58a8096  0.0s
 => => exporting attestation manifest sha256:8b1cb9b4513f153458a8f0d20b9cd6da612cfd72869a5  0.0s
 => => exporting manifest list sha256:40fc6a90d02dfcf7422f6e4fba63012bbf617181aa900b6f1921  0.0s
 => => naming to docker.io/library/waibase:vnext                                            0.0s
 => => unpacking to docker.io/library/waibase:vnext                                         7.0s
 => [development] resolving provenance for metadata file                                    0.0s
[+] Building 1/1
 ✔ development  Built                                                                       0.0s 
      204.12 real         1.10 user         0.98 sys
docker-compose -p waidev    up -d development
[+] Running 2/2
 ✔ Network waidev_default  Created                                                          0.1s 
 ✔ Container development   Started                                                          0.7s 
docker-compose -p waidev    exec development     bash || true
root@1e3ce10cd0c1:~/workspace# env
HOSTNAME=1e3ce10cd0c1
PWD=/root/workspace
HOME=/root
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35: [... cut ...]
img = /root/workspace/we2_image_gen_local/output_case1_sec_wlcsp/sign_formal_hx_layout_ota.bin
flash address = 0x29000, len of cur data = 167936
img = /root/workspace/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p1.img
flash address = 0xAB000, len of cur data = 700416
img = /root/workspace/we2_image_gen_local/output_case1_sec_wlcsp/inter_files/sign_formal_cm55m_s_application_p2.img
Output image: output_case1_sec_wlcsp/output.img
Output image: output_case1_sec_wlcsp/output.img

IMAGE GEN DONE
root@1e3ce10cd0c1:~/workspace# 
```

ashceniza, 2025-03-23:
@Tobyntobyn   can oyou have a look at the documentation on the doco branch, you can try running this, I've tested it locally and it works as in compile and generates an image, I need you to be  the guinea pig, Thanks mate, just updated with any errors so I can fix it.

ashceniza, 2025-03-23:
@Tobyntobyn   can you please test this, be the guinea pig.  You're good to go off the main branch this time, I've merged it.  Awaiting feed back on documentation.

Tobyntobyn, 2025-05-13:
I apparently like to ignore @ashceniza ;)
Docker repo is set up and ready to deploy to the team. Dev went through the process pretty seamlessly, we can pass it across to Adarsh next (it is unlikely he'll actually need to develop within the SDK, but useful for him to have access to all the moving parts of the project).
The post for the website is drafted and one step away from posting to the website

deva-wildlifeai, 2025-05-19:
Do we need to do anything else team? 
On my side, seems done but let me know if you need more testing :)

Tobyntobyn, 2025-05-20:
Hmm, it isn't deployed to everyone so is it considered done? I don't think it's been posted yet either.

karolinakisiel, 2025-06-10:
@wildlife-ai-adarsh i assigned this to you to do the final test and then we can close it. @Tobyntobyn, @ashceniza or @deva-wildlifeai would you be able to help Adarsh get across it? or point him to the right information please?

wildlife-ai-adarsh, 2025-06-14:
@karolinakisiel Will I need to use this. I will try to squeeze it in but won't be till after i get the Supabase stuff I need to finish done. Is that OK? Also is this needed for the app or just a test of the steup.

cc @ashceniza @Tobyntobyn

ashceniza, 2025-06-14:
@wildlife-ai-adarsh happy to help, as I created this monster I can help you sort this out if needed.  cc :@karolinakisiel @Tobyntobyn