wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #50: [Data Science] [Hardware] Brainstorm about creating the CamTrap DP format onto the PCB

deva-wildlifeai, 2025-05-13:
The goal of this task is to brainstorm on how you could recreate the [CamTrap DP format](https://camtrap-dp.tdwg.org/) on the SD Card. Here is a example of a [dataset](https://camtrap-dp.tdwg.org/example/). 

.
├── datapackage.json
├── deployments.csv
├── media
│   ├── 20210531082538-RCNX0031.JPG
│   ├── 20210531082538-RCNX0032.JPG
│   ├── 20210531082539-RCNX0033.JPG
│   ├── 20210531082539-RCNX0034.JPG
│   ├── 20210531082539-RCNX0035.JPG
│   ├── 20210531082540-RCNX0036.JPG
│   ├── 20210531082540-RCNX0037.JPG
│   ├── 20210531082540-RCNX0038.JPG
│   ├── 20210531082540-RCNX0039.JPG
│   ├── 20210531082541-RCNX0040.JPG
│   └── temp-folder
├── media.csv
└── observations.csv

deva-wildlifeai, 2025-05-13:
Could we?: 
1. create all the empty files in the mobile app when starting a deployment. 
2. send the files from the mobile app to the PCB 
3. make the pcb write into each file

deva-wildlifeai, 2025-05-15:
Here is the [report](https://github.com/wildlifeai/wildlife-watcher-testing/blob/main/_documentation/data_science_testing/addaxAI_testing/report_addaxAI_testing_WW500.md) on how we can use AddaxAI to get images annotated! 

### Feedbacks 
As it is, it's working: 
- we drag and drop the deployment folder
- we got a `result.csv` where one line = annotated image | bonding boxes | exif info from the image itself

Still few questions for brainstorming: 
1. Deployment information are coming from the mobile app, could the mobile app create, populate and upload on the device the `deployment.csv`?
2. Should we offer on the mobile app to record the output as a CamTrap DP formatted folder?

deva-wildlifeai, 2025-05-19:
Could we either?: 

## Create the CSV file during the deployment on the mobile app
1. Create all the empty files in the mobile app when starting a deployment. 
2. Send the files from the mobile app to the PCB 
3. Make the WW500 write into each file (dynamically or every 24h)

## *OR* Include a ZIP file in the mobile app that contains the necessary CSV
1. When the user is downloading the app, you have the CSV in it
2. When doing a deployment, it unzipping the content inside a temporary folder and then send it to the WW500
3. Make the WW500 write into each file (dynamically or every 24h)

victor-wildlife, 2025-05-20:
@deva-wildlifeai can we break down the files and find out if we really need them or we can leave empty? I agree most of the work should be done by the mobile app when the user Start Deployment.
I will frame the discussion as:

- datapackage.json : Does Adax AI need this file? If so, the mobile app should be able to create it with a minimum number of fields populated. 
- deployments.csv: The mobile app should be able to create this file and a single row with the deployment information. The WW500 should, every 24 hours, update the "deploymentEnd" date to the current date/time. @Tobyntobyn to explore how to set the auto "deploymentEnd".
- media: This folder and the filename convention (20210531082538-RCNX0031.JPG) are already implemented in the WW500, right? @Tobyntobyn 
- media.csv: Does Adax AI need this file? I am afraid this is the most problematic file, as we don't want the WW500 to constantly write the information for each image. Best to discuss this in our meeting, though.
- observations.csv: Does Adax AI need this file? If so, we can create an empty one as we don't want to track observations in this csv

acutetech, 2025-05-24:
I have not been following matters relating to these .csv files. I know they have been knocking around for some time. Can anyone tell me where I can learn about what is supposed to be in them, why they might be used etc? I just don't know.

I can make these observations:

1. It is my expectation that we will create a generic mechanism for bidirectional file transfer between the app and the phone.
2. As well as the .csv files (if necessary) I expect we might have files containing NN models and new firmware images for the AI processor. I expect there will be some configuration data that is best passed in a file. Images sent to the app as part of the installation process (checking the camera is pointing to the right target etc) will use this file transfer mechanism.
3. Just because a file passes between the device and the app does not mean that the file should be generated by the app, or parsed by the app. I would be reluctant to add functionality to the app unless it is essential.
4. For the same reason, I would be reluctant to add functionality to the device unless it is essential. It is already quite complex and file processing is quite slow. And C programming resource is very limited.
5. Remember also that the process of commissioning each device in the field, and decommissioning it, is a different process from that of planning and managing a deployment of multiple devices. I would expect that there are some tasks that are best done by a project manager, on a laptop, in the office, before and after devices are deployed in the field. Would originating and processing these .csv files be better treated as part of this planning and managing operation?

Tobyntobyn, 2025-05-26:
> * deployments.csv: The mobile app should be able to create this file and a single row with the deployment information. The WW500 should, every 24 hours, update the "deploymentEnd" date to the [...]
 - I'll look at this piece with Ash when we create the sequence diagram

> * media: This folder and the filename convention (20210531082538-RCNX0031.JPG) are already implemented in the WW500, right? [@Tobyntobyn](https://github.com/Tobyntobyn)
 - Yep, Albiet locally at the moment, but the new naming convention for files is: deploymentID_imageCount.jpg

karolinakisiel, 2025-06-10:
@deva-wildlifeai - is this completed? Can we close?

deva-wildlifeai, 2025-06-11:
Yes! 
As a conclusion, we'll use only this structure: 
.
├── deployment001
│ ├── deployments.csv
│ ├── media
│ │ ├── <timestamp>.JPG