wildlifeai/Seeed_Grove_Vision_AI_Module_V2 #56: AI processor's RTC does not update while in DPD

acutetech, 2025-07-02:
(This might be intermittent.)

I get the AI processor to enter DPD with a timer set to wake it. Typically after 10 minutes, in this example. The code prints the UTC time just before entering DPD and predicts the time at which it will wake. The timer works as expected, waking the AI processor after 10 minutes. But the AI processor then prints the time - which is the same as it was when the AI processor went to sleep.

Typically console print follows:

`
Entering DPD at 2025:07:02 04:26:58
Will wake at 2025:07:02 04:36:58
1st BL Modem Build DATE=Jan 17 2025, Version: 2.12
<snip>
**** WW500 MD. (WW500_A00) Built: 12:22:54 Jul  1 2025 ****
Wakeup_event = 0x0002, WakeupEvt1 = 0x0000 RTC Timer
<snip>
### Warm Boot ###
Woke at 2025:07:02 04:26:58
`
I have not spent much time trying to track this down. The BLE processor is notified that the AI processor has woken, and sends it the correct time. But this should not be necessary....

acutetech, 2025-07-02:
I note that, as currently written, when the BLE processor updates the AI RTC this takes several hundred ms, and this happens before the first picture is taken. We should ensure that RTC business does not delay taking photos.

victor-wildlife, 2025-12-04:
closing this as I imagine it is not an issue anymore, feel free to open it if it is though

acutetech, 2026-10-04:
The question is really whether we can reduce the time taken for the BLE processor to provide it with the correct date. I suggest this is deferred to #251 'Restructure the way state is saved across DPD to decrease time to the first photo'

acutetech, 2026-10-31:
From #245 (closed): the sleep timer is now corrected per board, but the AI processor's clock still
gains 2 to 4.6 % between updates from the BLE processor. Handle that here. One idea, in
`2026-10-27_XIP_save_state/CLAUDE_time_accuracy.md` (Problem 2): take the frame at once, then stamp and save it when the BLE processor has sent the time, as the photo already waits for the SD card.

acutetech, 2026-11-01:
Cause found, 1 November 2026 (branch 261101_AI_RTC).

The AI processor's RTC stops counting during deep sleep when no wake alarm is set, which means whenever timelapse is off (op 7 = 0). A camera that only wakes on motion loses all the time it spends asleep: in one bench run it was 45 s behind after 7 short sleeps. With an alarm set (op 7 > 0) the clock keeps running.

Fix, benched on one board (HM0360 image): when no alarm is set, leave the RTC0 interrupt unmasked in the power manager with its alarm interrupt turned off. The clock then keeps counting through the sleep, and nothing wakes the camera except motion or the BLE processor. To be made permanent in sleep_mode.c.

Still to fix:
 
- The first read after a wake can be stale. For part of the first second after a wake, reading the RTC can return the time the camera went to sleep (the Himax driver documents this in hx_drv_rtc.h). A photo stamped in that window gets the wrong time: 39 s early in one case. Next step: use the driver's synced read before the photo is stamped, or the BLE processor's time.

- Drift: the RTC still runs 2 to 4.6 % fast between setutcs (from #245).

acutetech, 2026-11-02:
Fixed by PR #284

closed 2026-11-02 by acutetech